heitzmann/gdspy #191: Is there any way i can attach label with a polygon?

mansigupta1998, 2021-07-26:


heitzmann, 2021-07-26:
No, polygons and labels are independent. You can attach properties to both, though.

mansigupta1998, 2021-07-26:
Thanks for your fast response.
Can you explain what all properties.
My problem statement is that,  after importing gds if I move any polygon, label attach to that polygon should move as well but not vice versa.
Also this attachment would help in various other things.

heitzmann, 2021-07-26:
There is no element attachment in the GDSII format. Labels are placed independently of polygons. If you move a label, only the label moves. The same goes for polygons. If you need a structure that moves together, place both the polygon and the label in a cell and move the cell reference around.

Properties are dictionaries associated to GDSII elements. You can access them through the [`properties`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.PolygonSet) variable.

mansigupta1998, 2021-07-29:
Ok thank you

closed 2021-07-29 by mansigupta1998