heitzmann/gdspy #194: Converting .gds file to ASCII format

nimeshpok, 2021-09-21:
OBJECTIVE - I want to use gdspy to read information contained in a .gds file and extract the mask data for a certain layer. 

ISSUE - I went through the online manual and the two examples - except for the small bit of information on loading files inside the "Loading a GDSII File" section I couldn't find much! I'd really appreciate your help?

Best wishes,
Nimesh

heitzmann, 2021-09-22:
Hello @nimeshpok 

What do you mean by “extract”? Will you save another file with only the desired layer? Is your layout hierarchical or only a single cell? Do you want to preserve the hierarchy? Or do you need a script that can be used with any sort of file in the future?

You will probably use [`Library.read_gds`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.read_gds) and [`Library.top_level`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.top_level) to find the cell (or cells) in the root of your hirearchy. From there, you can remove unnecessary elements with [`Cell.remove_polygons`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.remove_polygons) (similarly for paths and labels) from the root cell and its dependencies with [`Cell.get_dependencies(True)`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.get_dependencies) (or just remove from all cells in the library, regardless of the dependency tree). Then you can [save the remaining library](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.write_gds) in a new file, if that's what you want.

nimeshpok, 2021-09-22:
I really appreciate your reply Lucas. Thanks for your time.

For a simple case, let's say we have a single layer with a simple rectangular mask pattern. I was thinking of extracting the corners of this rectangular mask and writing the (x, y) coordinate to a text file. Is there a way to do this?

heitzmann, 2021-09-22:
One simple solution that will work for any type of mask (not just a rectangle) is to retrieve the bounding box of the layout:
```python
import numpy
import gdspy

gdsii = gdspy.GdsLibrary(infile="filename.gds")
main_cell = gdsii.top_level()[0]  # Assume a single top level cell
bbox = main_cell. get_bounding_box()

if bbox is None:
    print("Main library cell is empty.")
else:
    print(f"Bounding box: ({bbox[0, 0]}, {bbox[0, 1]}) - ({bbox[1, 0]}, {bbox[1, 1]})")
```

If you really want to inspect the points of the rectangle, you'll have to look for it in `main_cell.polygons` (`points = main_cell.polygons[0].polygons[0]` if there is only a single rectangle in the cell).

nimeshpok, 2021-09-22:
Thanks Lucas. That helped a lot!

Now, if we have another rectangular block inside the file say - 
```import gdspy

# The GDSII file is called a library, which contains multiple cells.
lib = gdspy.GdsLibrary()

# Geometry must be placed in cells.
cell = lib.new_cell('FIRST')

# Create the geometry (a single rectangle) and add it to the cell.
rect1 = gdspy.Rectangle((0, 0), (2, 1))
rect2 = gdspy.Rectangle((2.1, 0), (4.1, 1))
cell.add(rect1)
cell.add(rect2)
```
Can we get the bounding box for each rectangle? Or the corner coordinates?

Cheers,
Mesh

heitzmann, 2021-09-22:
Sure, you can use [`PolygonSet.get_bounding_box()`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.PolygonSet.get_bounding_box) for each element in `cell.polygons` (which is a `PolygonSet` list).

nimeshpok, 2021-09-22:
Thanks man. You're awesome 💯

heitzmann, 2021-09-22:
My pleasure!

closed 2021-09-22 by heitzmann