heitzmann/gdspy #195: coordinate precision issue after translate

clairetsai818, 2021-09-23:
Hi Developer,

I used [translate](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.PolygonSet.translate) function to move the PolygonSet. There's a coordinate precision problem when the dx or dy is negative.
The y-axis value before translate is 1.182, and dy is set to -2. However, the y-axis value became -0.8180000000001 after translation while I expected it should be -0.818. I've tried Decimal module which is often used to solve the precision issue of float numbers, but it doesn't work. Can you please give me some relavent advice? Thanks a lot!

Best Regards,
Claire

heitzmann, 2021-09-23:
@clairetsai818 gdspy uses floats to represent coordinates internally, and not all number can be represented exactly.
However, that should no be a problem for any real world application, since no fabrication tool with have precision (or accuracy) on the order of 10⁻¹³. Furthermore, when you save a GDSII file, [the coordinates are rounded to integers](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#units-in-gdsii) with the precision you specify, so this sort of representation problem won't matter in the end.

tvt173, 2021-09-23:
adding to @heitzmann's answer, you can use numpy's `round` method to round all the elements of an array to a specified precision. this is sometimes useful in this context to put your coordinates "on-grid" (to be consistent with the values which will be written to gds)

i.e.
```
poly_bounds = [[1.0231, 1.9390854], [4.3242, 3.2354325]]
np.round(poly_bounds, 3)
```
returns:
```
array([[1.023, 1.939],
       [4.324, 3.235]])
```

clairetsai818, 2021-09-24:
@heitzmann Got it! Thanks a lot for your explanation!
@tvt173 I think this function is definitely what I need. Thanks a lot!

closed 2021-09-24 by clairetsai818