heitzmann/gdspy #198: Duplicate Cell Error when reading from file

tylerflex, 2021-11-04:
Hi gdspy folks,

First, thank you for the very useful package.
I'm having some issues in the current version when reading and writing from file.
Here is a sample code with a `Library` containing one cell called `"Coupler"`.

```python
# write gds file
gds_path = 'file.gds'
if os.path.exists(gds_path):
    os.remove(gds_path)
lib.write_gds(gds_path)

print(lib.cells)
print("^ only one cell named 'Coupler'")
# read the gds from file
lib.read_gds(gds_path)
```
When I run this, the output is 
```python
{'Coupler': <gdspy.library.Cell object at 0x125cb3dc0>}
^ only one cell named 'Coupler'
╭──────────────────────────── Traceback (most recent call last) ────────────────────────────╮
│                                                                                           │
│ /var/folders/jx/9y0mtn3s3zzb6mzgmsw6s6gr0000gn/T/ipykernel_3749/4167473472.py:10 in       │
│ <module>                                                                                  │
│                                                                                           │
│ [Errno 2] No such file or directory:                                                      │
│ '/var/folders/jx/9y0mtn3s3zzb6mzgmsw6s6gr0000gn/T/ipykernel_3749/4167473472.py'           │
│ /usr/local/lib/python3.9/site-packages/gdspy/library.py:2635 in read_gds                  │
│                                                                                           │
│   2632 │   │   │   │   │   name = rename_template.format(name=record[1])                  │
│   2633 │   │   │   │   cell = Cell(name, exclude_from_current=True)                       │
│   2634 │   │   │   │   if name in self.cells:                                             │
│ ❱ 2635 │   │   │   │   │   raise ValueError("[GDSPY] Multiple cells with name: {0} in GDS │
│   2636 │   │   │   │   self.cells[name] = cell                                            │
│   2637 │   │   │   # STRING                                                               │
│   2638 │   │   │   elif record[0] == 0x19:                                                │
╰───────────────────────────────────────────────────────────────────────────────────────────╯
ValueError: [GDSPY] Multiple cells with name: Coupler in GDSII file
```

As you can see, there is only one cell named "coupler", yet I get this ValueError.

Am I doing something incorrectly?

Thank you!

tvt173, 2021-11-05:
hi @tylerflex, I believe the issue is that you are reading the GDS back into the existing library (which already has those cells). you should create a new library and read into that instead

tylerflex, 2021-11-05:
@tvt173 Thanks for your help, that solved it!

closed 2021-11-05 by tylerflex