heitzmann/gdspy #202: str' object has no attribute 'get_svg_classes'

poojabeniwal, 2022-01-12:
I am trying to convert gds file into svg file.
But facing error "str' object has no attribute 'get_svg_classes' ".
I was not able to understand how to resolve this issue, it would be great if you could help me to resolve this issue.

[image]

import gdspy
import numpy
import sys
import copy

import os 
os.environ['path'] += r';C:\Program Files\UniConvertor-2.0rc5\dlls'
import cairosvg

layout= gdspy.GdsLibrary(infile='C:\\Users\\pooja\\Downloads\\p1.gds')
cellList = layout.cells
cellList
for val in cellList:
        cell = cellList[val]
        cell.remove_paths(lambda path: path.layers[0] != val)
        cell.remove_labels(lambda label: label.layer != val)
        print(f"Processing {cell}")
        cell.write_svg('C:\\Users\\pooja\\svg\\AA_project_layout.svg')

heitzmann, 2022-01-13:
Thanks for the report. I've pushed a fix in the latest commit. Can you check if it works for you, @poojabeniwal?

poojabeniwal, 2022-01-13:
Thank you for your response.
Now I am getting another error.
[image]

heitzmann, 2022-01-13:
This error looks strange, because `ref` is assigned on the previous line. Can you share the GDSII file and the script that reproduces the error so that I can investigate the problem?

poojabeniwal, 2022-01-13:
When there are no references in gds (train.gds and test.gds) it works fine
But when there are references in gds (all other gds) it's throwing an error.

I have attached the script and the all gds.

[gds_to_svg.zip](https://github.com/heitzmann/gdspy/files/7863244/gds_to_svg.zip)

heitzmann, 2022-01-13:
The files you sent are working fine. Have you restarted your notebook kernel after reinstalling gdspy?

poojabeniwal, 2022-01-13:
Thank you, It's working now. I really appreciate your help.

closed 2022-01-13 by heitzmann