heitzmann/gdspy #205: How to get coordinates in one meta layer's polygon ??

etha-gh, 2022-02-04:
Hi !

I'm analyzing gds2 in my school,
Try to get one meta layer data, but I alway get all hole cell's data.

I hope you give me advice or sample code.

Best regards

import gdspy
import tripy
import pandas as pd
from plotnine import ggplot, aes, geom_polygon, geom_segment, geom_point

lib = gdspy.GdsLibrary(infile='test.gds')

###########################################################3
def convert_all(d2_array):
    ret_array = []
    for i in range(0, len(d2_array)):
        for j in range(0, len(d2_array[i])):
            ret_array.append(tuple(d2_array[i][j]))

    return ret_array

def convert_devide(d2_array):
    ret_array = []
    for i in range(0, len(d2_array)):
        ret_array2 = []
        for j in range(0, len(d2_array[i])):
            ret_array2.append(tuple(d2_array[i][j]))

        ret_array.append(ret_array2)
    
    return ret_array

def make_frame_from_triangles(triangles):
    x_start = []
    x_end = []
    y_start = []
    y_end = []
    for triangle in triangles:
        for i, pt in enumerate(triangle):
            next_index = (i + 1) % 3
            x_start.append(pt[0])
            x_end.append(triangle[next_index][0])
            y_start.append(pt[1])
            y_end.append(triangle[next_index][1])
    df = pd.DataFrame({'x': x_start, 'y': y_start, 'xend': x_end, 'yend': y_end})
    return df
########################################################################################

cells_polygons = []
cells_polygons.append(convert_devide(lib.cells['FlexPath1'].get_polygons()))

triangles = []
for j in range(0, len(cells_polygons)):
    for i in range(0, len(cells_polygons[j])):
        triangles.append(tripy.earclip(cells_polygons[j][i]))    
    
all_tri = []
all_tri = convert_all(triangles)

df = make_frame_from_triangles(all_tri)
print((ggplot(df, aes(x='x', y='y')) + geom_segment(aes(x='x', y='y', xend='xend', yend='yend'))))

heitzmann, 2022-02-04:
@etha-gh You should be able to get the polygons in a single layer/datatype with [`Cell.get_polygons`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.get_polygons), for example:

```python
layer = 3
datatype = 4
polygon_list = cell.get_polygons((layer, datatype))
```

etha-gh, 2022-02-04:
Thank you so much!
I try to do it.

etha-gh, 2022-02-04:
Thank you reply!
I try to run sample you give me, but it says nothing.

if there are wrong?
I'm so sorry take your time for me.

import gdspy

lib = gdspy.GdsLibrary(infile='test.gds')
cell = gdspy.Cell('test.gds')

layer = 3
datatype = 4
polygon_list = cell.get_polygons((layer, datatype))

print(polygon_list)

heitzmann, 2022-02-04:
You have to load a cell from your library. This line:
```python
cell = gdspy.Cell('test.gds')
```
creates an empty cell with the name `'test.gds'`

etha-gh, 2022-02-05:
So sorry, I'm new come.
just started few weeks ago. 

I'm really really sorry.

`import gdspy

lib = gdspy.GdsLibrary(infile='test.gds')

layer = 12
datatype = 4
polygon_list = cell.get_polygons((layer, datatype))

print(polygon_list)`

It says 'cell' is not defined, 
how do I can defined cell?

heitzmann, 2022-02-06:
@etha-gh I looks to me that you should be first going through a few Python tutorials to learn the language, if it is new to you. Then, you have to read through gdspy's [documentation](https://gdspy.readthedocs.io), because you seem to be guessing what to do.

Meanwhile, you probably need to do something like this (`'cell_name'` is the name of the cell you want to extract from the GDSII file):
```python
import gdspy

lib = gdspy.GdsLibrary(infile='test.gds')

cell = lib.cells['cell_name']

layer = 12
datatype = 4
polygon_list = cell.get_polygons((layer, datatype))

print(polygon_list)

# Coordinates from first polygon in the list
print(polygon_list[0])
```

etha-gh, 2022-02-06:
[test_gds.py.zip](https://github.com/heitzmann/gdspy/files/8010449/test_gds.py.zip)

Thanks for your advice.
I'm study about GDSII files. and make one meta layer's graphic.
here is my all code I sent to you in zip file.

If you find something wrong, I hope you tell me.

So thank you give me advice and lecture.
I will read your Document and keep to study.

Best Regards

closed 2022-02-09 by etha-gh