heitzmann/gdspy #206: Module not found error

anujashok194, 2022-02-11:
[image]
[image]

Can you please help me with this?

heitzmann, 2022-02-11:
Did you install gdspy on the same virtual environment you're trying to use it from? On the Command Prompt where you installed gdspy, does the following command work: `python -c "import gdspy"`?

anujashok194, 2022-02-11:
No its not working

heitzmann, 2022-02-11:
How did you install it the first time? On Windows, gdspy need a properly configures development environment for compilation of C extensions. If your environment is not configured correctly, it will appear to have been installed but it won't be. If that's the case, uninstall it with pip (as you did the first time) or manually remove all files from site-package and use one of the provided wheels (_*.whl_) in the [releases page](https://github.com/heitzmann/gdspy/releases/tag/v1.6.11).

anujashok194, 2022-02-11:
[image]

Thank you. It work.

closed 2022-03-02 by heitzmann