heitzmann/gdspy #207: About the gdspy.Text() class

ru-tan, 2022-04-02:
Dear developers of Gdspy,

I am a user of gdspy from the integrated photonics community. Gdspy has been a powerful and essential tool for me, but one issue has bothered us.

By default, the gdspy.Text() creates polygons with only horizontal or vertical edges, but some of the polygons share the same vertices, as shown below. We are told by some silicon photonics foundries that this violates their design rules.
[image]

Currently, this issue can be solved first by using gdspy.Label() and then converting the label into a PCell in Klayout (shown below), but it is not so convenient. 
[image]

Could you consider providing additional font styles to address this issue?
Thank you very much!

tvt173, 2022-04-04:
hi @ru-tan ,

I also work in photonics and contribute to both gdspy and [gdsfactory](https://github.com/gdsfactory/gdsfactory) (which is built on top of gdspy). The gdsfactory `text()` function may be better suited for your needs. That said, I usually use custom fonts and realize it is not super straightforward to do so in either library. I created an [issue](https://github.com/gdsfactory/gdsfactory/issues/259) in gdsfactory for this

@joamatab

heitzmann, 2022-04-04:
Hello @ru-tan, font support is a very deep rabbit hole I wouldn't have time to work on.

That would require at least:
- Support the most common formats
- Support for system-specific font paths/loading libraries
- Plus implementation “detail”: font metrics, kerning, ligatures, encoding, writing directions, etc.
The solution here is probably to depend on other libraries, but that increases the complexity of the build system in all platforms. I might be able to work on that in the future, but right now this is a very low priority.

Creating another “font” in the same way I did with the current one might solve the problem with a specific foundry, but there would always be processes with different design rules…

Meanwhile, you can try using an [offset](https://heitzmann.github.io/gdstk/geometry/gdstk.offset.html) on your text to avoid those 4-way intersections or follow the example on [how to use system fonts](https://heitzmann.github.io/gdstk/how-tos.html#system-fonts) through [matplotlib](https://matplotlib.org/).

tvt173, 2022-04-05:
from my experience, it's smart to be dumb with fonts 😆 

as you say, @heitzmann , it can be a never-ending rabbit hole. we should not try to implement every font you have in Microsoft Word, for example, or get too caught up in all the complexities of actual fonts in a word-processor. but we should have

A. A reasonable default font (or maybe a couple, as the definition of reasonable might vary!)
B. Capability to load a custom character set... which I would argue should be contrained to only have character geometries and fixed character widths and spacings. Going much further beyond this provides little value to polygon text. At the end of the day, all you want is for the characters to be manufacturable and readable. Doesn't need to be fancy.

but anyways, then you start to get into a fairly opinionated design, which seems a better fit for gdsfactory to me than gdspy/gdstk (which sticks more to the essentials)

or as you say, a reasonable, simple alternative (if it works for you) is to simply add an offset to the existing characters in gdspy

tvt173, 2022-04-15:
hi @ru-tan , apparently this is already built into phidl, which is a library built on top of gdspy.
see: https://github.com/gdsfactory/gdsfactory/issues/259#issuecomment-1099768545

thanks to @joamatab and @spauka for sharing

amccaugh, 2022-06-14:
Yes, in [phidl](https://github.com/amccaugh/phidl/) the default font used is DEPLOF (originally made by David Elata at Technion to avoid issues like this, and also prevent any floating islands which causes issues when doing certain processes like MEMS). Thanks to @spauka, it also recently has full truetype (font support)[https://phidl.readthedocs.io/en/latest/geometry_reference.html#Text]

You're obviously welcome to use the phidl text function (which as mentioned is built on top of gdspy), but if you're looking for something a little more quick-and-dirty, I'd recommend just grabbing the individual character glyphs for the DEPLOF font here: https://github.com/amccaugh/phidl/blob/master/phidl/constants.py

closed 2022-10-01 by heitzmann