heitzmann/gdspy #208: get_labels with rotated references

joamatab, 2022-05-09:
How can we make sure that get_labels returns labels with the correct rotation?

For example

```
if __name__ == "__main__":
    # add a child cell with a label
    c1 = gdspy.Cell("child")
    label = gdspy.Label("label", (0, 0))
    p = gdspy.Polygon(((0, 0), (1, 0), (0, 1)))
    c1.add(p)
    c1.add(label)

    # add parent with rotated cell
    rotation = 90
    c2 = gdspy.Cell("parent")
    c1ref = gdspy.CellReference(ref_cell=c1, rotation=rotation)
    c2.add(c1ref)
    assert c2.get_labels()[0].rotation == rotation

```

joamatab, 2022-05-10:
Hi Lucas,

How about we add a new function that returns a rotated label?

joamatab, 2022-05-10:
See https://github.com/gdsfactory/gdsfactory/pull/373

tvt173, 2022-05-10:
@joamatab , out of curiosity, what's the motivation for this?

if we were to add this to gdspy, maybe it could be a global flag? it seems you would always want it to behave one way or the other. if the global flag is set, rotate the labels (consistent with gdstk behavior). if not, just leave them as-is (backwards compatible)

heitzmann, 2022-05-10:
@tvt173 A flag can be used to maintain compatibility indeed!
@joamatab I've included a possible solution in the last commit. Would you mind checking to see if it works properly? I don't have the time to write extensive tests for it now, I've just ported from gdstk, but their internals are a bit different.

joamatab, 2022-05-10:
This worked great Lukas, maybe we could make a new release

Hi Troy,

the motivation is that I've been documenting different test protocols in gdsfactory
https://gdsfactory.github.io/gdsfactory/notebooks/09_mask.html

The default method included in gdsfactory labels each fiber port, and that one does not have issues

However the two other methods documented use a single label per device, and the label contains relative port positions. However, that relative port positions are not correct when the subdie is rotated, and if we don't keep track of subdies transformations it is not possible to correct afterwards

let me know what you think
Joaquin

joamatab, 2022-05-10:
Here is a small fix
https://github.com/heitzmann/gdspy/pull/209

joamatab, 2022-06-06:
Thank you for the latest release Lucas

closed 2022-06-06 by joamatab