heitzmann/gdspy #213: polygon,path layer and datatype property is wrong

cecial, 2022-06-02:
Hi, polygon.layer and polygon.datatype is not working from the API reference, instead, i can use polygon.layers[0] and polygon.datatypes[0]. 
same as path.

heitzmann, 2022-06-02:
@cecial Thanks for reporting. Could you be more specific where in the documentation is the error you found?

cecial, 2022-06-02:
> @cecial Thanks for reporting. Could you be more specific where in the documentation is the error you found?

Hi, looks i misunderstand the API reference in https://gdspy.readthedocs.io/en/stable/reference.html#polygon
Polygon can be created with layer and datatype, so when i want to access the layer and datatype, i use polygon.layer and polygon.datatype, which is a straightforward thought, but actually, i need use polygon.layers[0] and polygon.datatypes[0]

BTW, is it possible that a polygon has multible layers and datatypes? if not, why don't we use polygon.layer and polygon.datatype?  

in the API reference, there is no description about how to access the layer and datatype for polygon.
I think polygon is a PolygonSet, in which description about  the layers and datatypes are there, i think that may be the reason
and a polygonset can have multible layer and dataype?

cecial, 2022-06-02:
> That's correct, `Polygon` has `PolygonSet` as a base class, that's why the attributes are the lists `layers` and `datatypes`. The reason for using sets is that many operations only make sense when operating or returning sets of polygons, so that choice of base class unifies the API. This is deeply-rooted in how gdspy was conceived, and it has been properly reworked in [gdstk](https://github.com/heitzmann/gdstk/).

Hi, Thanks for  this quick replay, i will try to look gdstk

cecial, 2022-06-02:
issue solved

closed 2022-06-02 by cecial