heitzmann/gdspy #216: how to do y_reflection with cell view.

jamesvitali, 2022-09-28:
there is a param named "x_reflection" in gdspy.CellReference , but no method for the reflection to Y axis. If I would like to reflect the cell but not increase the final gds size , any way I can take? thanks

tvt173, 2022-09-28:
hi @jamesvitali, handy trick to remember-- a `y_reflection` is equivalent to an `x_reflection` + a 180 degree rotation!

jamesvitali, 2022-09-30:
hi @tvt173 thanks for your help , but for cell, any way to do mirror over a line through points 1 and 2? thanks

tvt173, 2022-09-30:
hi @jamesvitali , i recommend you consider using a higher-level library, such as gdsfactory or phidl. they are both based on gdspy and have such a function as you are requesting. i don't believe gdspy has such a function.

I give examples for gdsfactory as that's what I use primarily:

API reference:
https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.ComponentReference.html#gdsfactory.ComponentReference.mirror

Example:
https://gdsfactory.github.io/gdsfactory/notebooks/02_movement.html
[image]

or of course you can also just copy the code out of the mirror function in gdsfactory and use it if you prefer to stay in pure gdspy

heitzmann, 2022-10-01:
@jamesvitali Any mirror operation can be accomplished by an x-reflection + rotation. If I'm not mistaken, to mirror along a specific line, you can do an x-reflection and rotate by twice the angle define between the axis and the x-axis.

closed 2022-10-01 by heitzmann