heitzmann/gdspy #217: any way to reduce the gds file size

jamesvitali, 2022-10-05:
The size of gds is too big that is over 10G , is any compressed method to reduce the size? thanks

tvt173, 2022-10-05:
for a gds that already exists, you can zip it (many programs can open compressed gds files as-is) or convert to oasis format (klayout can do this for you). if you're making a gds from scratch, try making better use of hierarchy. that's typically the best way to keep gds file size down, when possible

jamesvitali, 2022-10-05:
thanks for your suggestion, so if from scratch, that means I need to find the smallest unit that has the same pattern to construct the gds, is that correct

tvt173, 2022-10-05:
yes, exactly... build `Cell`s and construct them hierarchically with `CellReference`s

jamesvitali, 2022-10-05:
thanks @tvt173  any tool or method you know can help to get the similar pattern if I have a large array?

tvt173, 2022-10-05:
yeah, use a [CellArray](https://gdspy.readthedocs.io/en/stable/reference.html#cellarray)

jamesvitali, 2022-10-05:
let's say that i have a big array and I don't know what pattern it has. But need to reduce the file size. Per our discussion, I need to find the similar pattern out and construct them hierarchically with CellReferences. but any tool or method I can use to find the pattern from the array?

closed 2022-10-11 by jamesvitali