heitzmann/gdspy #218: stuck in hierarchically with CellReferences

jamesvitali, 2022-10-11:
target to construct gds hierarchically with CellReferences but it gets infinite loop when call CellReferences, the way I use is not correct?

for i in range(num):
       ...................
       index = round(step[i]/10)
        rect = gdspy.Rectangle((.....))
        if "cell_" + str(index) in globals():
            globals()["ref_" + str(index) + str(i)] = gdspy.CellReference(globals()["cell_" + str(index)],origin=(x,y))
            globals()["cell_final_" + str(index)].add([globals()["ref_" + str(index) + str(i)]]) <==it causes  infinite loop
        else:
            globals()["cell_" + str(index)] = lib.new_cell(str(index))
            globals()["cell_final_" + str(index)] = lib.new_cell(str("cell_final_"+str(index)))
            globals()["cell_" + str(index)].add(rect)
            globals()["ref_" + str(index) + "_0"] = gdspy.CellReference(globals()["cell_" + str(index)])
            globals()["cell_final_" + str(index)].add([globals()["ref_" + str(index) + "_0"]])
         ...............................
for i in range(num):
        ....................
        final.add(globals()["cell_final_"+str(index)])

tvt173, 2022-10-11:
hmm this is some messy code with a lot going on...
I recommend you start with the example here and slowly modify it towards what you need
https://gdspy.readthedocs.io/en/stable/gettingstarted.html#references

jamesvitali, 2022-10-12:
thanks @tvt173 
now it can work but still have 2 questions:
1. in the example you recommend, it uses "gdspy.Cell" to create a cell, but the result is empty; however, if use "lib.new_cell" instead (not only the first cell), the result can be seen with klayout, do you know why?
2. now I use "lib.new_cell" to create cells (around 16), and each cell has its own references. I found the gds size will go larger along with the number of references. is that normal? I thought the size only depends on the number of cell you created, am I wrong?

Cell ("cell_final_5", 0 polygons, 0 paths, 0 labels, 2188 references)
Cell ("cell_final_6", 0 polygons, 0 paths, 0 labels, 4752 references)
Cell ("cell_final_7", 0 polygons, 0 paths, 0 labels, 6492 references)
Cell ("cell_final_8", 0 polygons, 0 paths, 0 labels, 7508 references)
Cell ("cell_final_9", 0 polygons, 0 paths, 0 labels, 10604 references)
Cell ("cell_final_10", 0 polygons, 0 paths, 0 labels, 10856 references)
Cell ("cell_final_11", 0 polygons, 0 paths, 0 labels, 14060 references)
Cell ("cell_final_12", 0 polygons, 0 paths, 0 labels, 16192 references)
Cell ("cell_final_13", 0 polygons, 0 paths, 0 labels, 19520 references)
Cell ("cell_final_14", 0 polygons, 0 paths, 0 labels, 20820 references)
Cell ("cell_final_15", 0 polygons, 0 paths, 0 labels, 21456 references)
Cell ("cell_final_16", 0 polygons, 0 paths, 0 labels, 20508 references)
Cell ("cell_final_17", 0 polygons, 0 paths, 0 labels, 19752 references)
Cell ("cell_final_18", 0 polygons, 0 paths, 0 labels, 19080 references)
Cell ("cell_final_19", 0 polygons, 0 paths, 0 labels, 20516 references)
Cell ("cell_final_20", 0 polygons, 0 paths, 0 labels, 2200 references)

tvt173, 2022-10-12:
1. I believe `lib.new_cell()` first creates the cell, then adds it to the library. If you are creating a cell with `gdspy.Cell()` instead, you should be sure to write it to a library afterwards. Then it is that _library_ that you write to gds.
2. References do have a small memory footprint also. In your example, you appear to have hundreds of thousands of references, so I would still expect a somewhat considerable size, though it should still be smaller than a flattened cell, as long as each of those references is more complex than i.e. a simple rectangle. You may be able to drive down size further by using CellArrays instead.

closed 2022-10-13 by jamesvitali