heitzmann/gdspy #221: the direction of polygon in the cellreference is wrong

jamesvitali, 2022-10-26:
if we have many polygons with different shape/direction in the first first quadrant, we want to have the same polygons in the second quadrant (opposite position but the same shape );however, if we use mirror/reflection or rotation on the cellrefence, though the position of polygon in the second quadrant is what we expect but the direction of polygon in the second quadrant is wrong. more precisely, the direction is mirror in the second quadrant not the same with the first quadrant. any good way to solve it on the cellreference?

heitzmann, 2022-10-28:
@jamesvitali If you use reflection, your shapes will be flipped. If you don't want that, you have to translate and rotate only.

jamesvitali, 2022-10-31:
thanks for your comment;however , if i already have the cellreference and don't want to new cell to increase its size, any good idea ? cellreference can do the rotation but it changes the angle of polygon as well.
 for example : current cellreference:====// // // and the target is // // //========// // //

heitzmann, 2022-11-01:
I can't see your example, but from what you're saying, I don't think there are other alternatives.

closed 2022-11-04 by jamesvitali