heitzmann/gdspy #222: may be it is a bug for GdsLibrary.new_cell()

mujingfatasy, 2022-11-24:
Hello, I'm trying to using gdspy to inductors in large quantities, but the gdspy can only work once,and failed at the second time. The error was triggered by GdsLibrary.new_cell() and look like this:
"ValueError: [GDSPY] Cell named turn1 already present in library."
then I refered to the API documentation and found "new_cell(name, overwrite_duplicate=False, update_references=True)"(https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.new_cell)， but even I changed "overwrite_duplicate" parameter to True, the error appeared still. 
And then I found that new_cell function will first call Cell class to initialize a cell and then pass the cell to GdsLibrary.add() function where the overwrite_duplicate parameter has been passed，but when Cell class initialize a cell，it will also call GdsLibrary.add() and overwrite_duplicate parameter was not given and then will cause a problem

mujingfatasy, 2022-11-24:
by the mean, the version of gdspy I used is 1.6.12

tvt173, 2022-11-24:
yeah, I'd agree that seems to be a bug. it should be fixed though (and better practice) if you first instantiate a new library and add your cell to that, rather than using `gdspy.current_library` directly (which it seems you are doing)

for example
```python
lib = gdspy.Library(name="mylibrary")
lib.new_cell(name)
```

mujingfatasy, 2022-11-24:
> OK, thanks a lot, but it seems that I was doing the same way as your recommendation，here is my source code :
   lib=gdspy.GdsLibrary(name="stacked_balun")
   turn1=lib.new_cell("turn1")
gdspy.current_library seems to be called by “__init__” function of Cell class which is call be lib.new_cell

tvt173, 2022-11-24:
hmm, interesting. to be honest, i don't usually use the `new_cell` function. I would usually create the cell first then add it like `lib.add(cell)`. I would give that a try. if you still get the problem, please share a reproducible example. or try submitting a patch. please by aware that you shouldn't add a cell with the same name twice, due to limitations of the GDSII format.

and if you are just getting started with gdspy, i would consider trying [gdstk](https://github.com/heitzmann/gdstk) instead, the successor to this project

heitzmann, 2022-11-25:
@mujingfatasy This seems to be a problem with the global library maintained internally (used in early versions of gdspy).
You can try to set `gdspy.library.use_current_library = False` right after importing gdspy to fix the issue.

mujingfatasy, 2022-11-26:
> @mujingfatasy This seems to be a problem with the global library maintained internally (used in early versions of gdspy). You can try to set `gdspy.library.use_current_library = False` right after importing gdspy to fix the issue.

OK，thanks a lot

mujingfatasy, 2022-11-26:
OK，thanks a lot