heitzmann/gdspy #226: Use gdspy in Klayout Macro developement (reopened)

atofini, 2023-01-12:
Reopening the issue #139 

Can you elaborate how the pre-compiled binary can be ran in KLayout? I get a circular path error when I use it.

Error code: 
cannot import name 'clipper' from partially initialized module 'gdspy' (most likely due to a circular import) (C:\Users\Alex.tofini\AppData\Roaming\KLayout\lib\python3.8\site-packages\gdspy\__init__.py)
  C:\Users\Alex.tofini\AppData\Roaming\KLayout\lib\python3.8\site-packages\gdspy\polygon.py:39

lucas-flexcompute, 2023-01-20:
You have to install the pre-compiled version in your KLayout python environment. I don't use Windows, so I can't give you exact directions. In Linux, Klayout uses the system python by default, so every installed package is immediately available. It looks like on Windows you'll have to select the correct python path to install the [whl](https://github.com/heitzmann/gdspy/releases).

heitzmann, 2023-01-20:
I don't use Windows, so I can't tell exactly what to do, but you have to use the whl installers from the releases and install them using the python path from Klayout (I believe it ships with its own python on Windows).