heitzmann/gdspy #229: GdsLibrary object has no attribute 'cells'

tinaa160, 2023-05-23:
Hi, 

Really appreciate for creating such useful module.

I encounter a AttributeError: 'GdsLibrary' object has no attribute 'cells' by running the following code:

```
lib = gdspy.GdsLibrary(name = "one")
lib.read_gds('xxx.gds')
for c in lib.cells:
    print(c)
```

The code was okay at first. (using v1.6.11)
Then I install v1.6.12 on my another laptop which run with this error.
I also tried to install v1.6.11 but it turns out the result was same.

heitzmann, 2023-05-23:
This is very strange. Do you have any files or directories named gdspy in your path?

Can you try the following?
```python
import gdspy
lib = gdspy.GdsLibrary(name="one")
print(lib.cells)
```

Then with your file:
```python
import gdspy
lib = gdspy.GdsLibrary(infile="xxx.gds")
print(dir(lib))
```

tinaa160, 2023-05-24:
Hi,

I just realize there was an installation error like the following message:

---------------------------
TEST FAILED:<my installation path> does NOT support .pth files error: bad install directory or PYTHONPATH

You are attempting to install a package to a directory that is not on PYTHONPATH and which Python does not read ".pth" files from. 
The installation directory you specified (via --install-dir, --prefix, or the distutils default setting) was: 
 
and your PYTHONPATH environment variable currently contains: ........
---------------------------

After specifying the correct PYTHONPATH, the code can work!

Thanks a lot for your help!

closed 2023-05-24 by tinaa160