heitzmann/gdspy #230: Multiple cells with name: FGCCTE in GDSII file

dany197, 2023-05-27:
Hi, I have been using gdspy since 2019.  This is very useful and thanks for your effort.

I recently got a problem when switch the same code from old (win 10)  to new computer (win 11). In this new computer, I have python(3.10.9), gdspy (1.6.13).  The problem comes out when I combine multiple gds files to one big gds file.  The code I use is attached below.

#####################
lib = gdspy.GdsLibrary()
main = lib.new_cell('main')
ref = []

dev1 = lib.read_gds(infile ='dev1.gds').cells['dev1_main']
dev2 = lib.read_gds(infile ='dev2.gds').cells['dev2_main']

ref.append(gdspy.CellReference(dev1, (0, 0), magnification=1, rotation=0))
ref.append(gdspy.CellReference(dev2, (0, 0), magnification=1, rotation=0))

main.add(ref)
lib.write_gds('main.gds')

#####################
This code works fine on my old computer. But raise ValueError("ValueError: [GDSPY] Multiple cells with name: FGCCTE in GDSII file") on my new computer.  FGCTE is a basic subcell which is reused in both dev1 and dev2. Could you help have a look and any suggestions will be appreciated!! Thanks a lot !!!

heitzmann, 2023-05-27:
Instead of reading into the same library, you can [`add`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.add) individual cells and overwrite any duplicates (assuming cells with the same name in both libraries are identical, there should be no problem), for example:

```python
lib1 = gdspy.GdsLibrary(infile ='dev1.gds')
dev1 = lib1.cells['dev1_main']

lib2 = gdspy.GdsLibrary(infile ='dev2.gds')
dev2 = lib2.cells['dev2_main']

lib = gdspy.GdsLibrary()
lib.add(dev1)
lib.add(dev2, overwrite_duplicate=True)

main = lib.new_cell('main')
main.add(gdspy.CellReference(dev1, (0, 0), magnification=1, rotation=0))
main.add(gdspy.CellReference(dev2, (0, 0), magnification=1, rotation=0))

lib.write_gds('main.gds')
```

dany197, 2023-05-27:
It works now. Thank you so much, Heitzmann! You are super brilliant！！