heitzmann/gdspy #231: result of get_bounding_box() for a polygon is wrong

tinaa160, 2023-05-30:
Hi,

I would like to obtain the bounding box of a polygon(108:250)  in a cell.
polygon(108:250) is a rectangle polygon.

Here's how I did:
`
all_pol = top_cell.get_polygons(by_spec = True)
target_layer = all_pol[108, 250]
print(target_layer)
tmp_target_layer = gdspy.Polygon(target_layer)
out = tmp_target_layer.get_bounding_box()
print(out)
`
output of "print(target_layer)"
[[15, 15],
[15, 45],
[45, 45],
[45, 15]
]

output of "print(out)"
[[15, 15],
[**15**, 45]
]

It seems output of "print(out)" should be 
[[15, 15],
[**45**, 45]
]

Do I misunderstand the usage of the function?

Thanks a lot,

heitzmann, 2023-05-31:
Your polygon creation is wrong. If you print `target_layer` you'll see that the output is not what you're showing. It should be like this:
```python
In [1]: import gdspy

In [2]: top_cell = gdspy.Cell("CELL").add(gdspy.Rectangle((15, 15), (45, 45), layer=108, datatype=250))

In [3]: all_pol = top_cell.get_polygons(by_spec=True)

In [4]: target_layer = all_pol[108, 250]

In [5]: print(target_layer)
[array([[15, 15],
       [15, 45],
       [45, 45],
       [45, 15]])]
```
You see: `all_pol` is a dictionary of **lists** of polygon vertices, so, if you want to create a new polygon, you have to select one of the list items:
```python
In [6]: tmp_target_layer = gdspy.Polygon(target_layer[0])

In [7]: out = tmp_target_layer.get_bounding_box()

In [8]: print(out)
[[15 15]
 [45 45]]
```

tinaa160, 2023-06-01:
Hi heitzmann, 
Yes your are correct!
It works after I use the function correctly.
Thanks a lot

closed 2023-06-01 by tinaa160